willyvmm/mouSTer #2: Only one button supported for C64 mouse

ssuukk, 2021-12-01:
The Portal game on C64 requires two mouse buttons, one is for blue portal, another for pink. It seems mouSTer supports only one mouse button in C64 mode, though!

willyvmm, 2021-12-01:
It's included in the new upcoming version.

willyvmm, 2023-01-30:
Try this Firmware:  https://we.tl/t-NqNx1OknEs

ssuukk, 2023-01-31:
I don't remember how to update the fw, but if it is enough to put it in the root of some SD drive and turn mouster on, then it didn't help (or didn't update).

radius75, 2023-01-31:
@ssuukk 
Clean the flash drive.
Create a MOUSTER folder
Put the FW file in the folder.
Plug in the Mouster with a flash drive and power the C64
Wait a minute.
Delete the FW file from the flash drive.
Plug it back in and power up the C64
Check the FW version in the automatically created INI file.

ssuukk, 2023-01-31:
działa

willyvmm, 2023-02-07:
Thanks for Feedback.

closed 2023-02-07 by willyvmm