willyvmm/mouSTer #5: C64: Question about 3xFire Joy

radius75, 2021-12-27:
[image]
It is possible to setup a mouster to have a C64Joy 3xFire with a PC Joypad
Any sugestion for configuration?
Thanks

radius75, 2022-01-04:
It's already working for me.
I set the buttons like this.
Maybe it will be useful to someone. (do not close thread)

For standard PC XBOX360 Gamepad

A = Fire1
B = Fire2 (potx)
X = Fire3 (poty)
Y = AutoFire1

```
[mouster]
mode=gpad
[gamepad]
number=00
revpotlines=true
gbutton01=mb
gbutton02=fire
gbutton03=rb
gbutton04=*fire
gbutton05=none
gbutton06=none
gbutton07=none
gbutton08=none
gbutton09=none
gbutton10=none
gbutton11=none
gbutton12=none
gbutton13=none
gbutton14=none
gbutton15=none
gbutton16=none
```

willyvmm, 2022-01-06:
That's the correct attempt to have 3 buttons working on most 8/16 bit machines.

radius75, 2022-01-07:
There is a problem with reading potX and potY in gpad mode.
To run them at all, you must first mouse and then gpad in ini file.

Reproduction step:

- run c64 + mouster
- insert pendrive, ini mode = mouse, 1351
- power off, unplug pendrive
- power on, test mouse ok
- insert pendrive mode = gpad
- 3xbutton joy works now on gamepad

If we do power on / off now, fire2 and fire3 stop working.
You have to repeat the whole procedure from the beginning.
Otherwise I can't get mouSTer to make this work.

Jumpman64, 2023-02-01:
Would be really nice to have three buttons for supported games. 
Tried this setup for the first time. It still didn't work with these gamepad settings in firmware 3.17.3377:
mode=gpad
revpotlines=true
gbutton01=mb
gbutton02=fire
gbutton03=rb
gbutton04=*fire

i'm using a DS4 controller + 8bitdo wireless adapter. 
X works as fire, triangle as auto fire. Square and circle do nothing.
Tested with [Joyride 1.6 ](https://csdb.dk/release/?id=219494) on an U64

radius75, 2023-02-02:
With FW 3.17.3377 I couldn't get Fire 2 and 3 to work either.
C64c

willyvmm, 2023-02-07:
Hi. 
Please try the latest firmware: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3470
This option has been removed by me by mistake and is restored in this build.

radius75, 2023-02-07:
Fire 2 i 3 (potx i poty) działają poprawnie, a oba te przyciski działają nadal nawet po zrobieniu restartu zasilania.

Przydało by się żeby Mouster sam wykonywał reboot zaraz po zaczytaniu configu z INI. (bez zrobienia tego, załadowana konfiguracja nie uruchomi się) Oszczędzi to często czas na wyłączanie zasilania C64 a często wgrywanie programu/gry od nowa.
--
Fire 2 and 3 (potx and pots) work fine and both of these buttons still work even after doing a power reboot.

It would be useful for Mouster to reboot itself right after reading the config from INI. (without doing this, the loaded configuration will not start) This will often save time for turning off the C64's power and often reloading the program / game.

radius75, 2023-02-07:
Jeśli nie stworzy to nowego problemu, nieskończonej pętli ładowania INI i Restartu.
Może jeśli się da to restart np zawsze po usunięciu pendrive USB

--
If it doesn't create a new problem, infinite INI load loop and Restart.
Maybe if it is possible to restart, e.g. always after removing the USB flash drive

radius75, 2023-02-07:
Bug naprawiony, wątek do zamknięcia.

--
Bug fixed, thread to be closed.

willyvmm, 2023-02-08:
> If it doesn't create a new problem, infinite INI load loop and Restart.
> Maybe if it is possible to restart, e.g. always after removing the USB flash drive

Best ideas are the simplest and most difficult to find.
I struggled with this issue for a really long time, and could not find any good solution. But the solution is so simple :)
Thanks for the Idea. It is implemented in the latest Pre-Release 3.17.3475 https://github.com/willyvmm/mouSTer/releases/tag/3.17.3475

Jumpman64, 2023-02-20:
tested mouSTer.fw.3.17.3475  - Mario is a blast with 2nd button mapped to square on a PS4 pad.
Great work!

willyvmm, 2023-02-22:
Thanks for the feedback.

closed 2023-02-22 by willyvmm