willyvmm/mouSTer #8: CD32 mode issue

DigiTurk84, 2022-01-06:
Greetings,

I recently bought two mouSTers, to use on a C64, an A1200 and a CD32.

I tested both mouSTers with latest firmware (3.16.2600) on the A1200 and the CD32 with the following outcome:

- Mouse is working perfectly
- Joystick mode is working perfectly, fire, auto fire and second button
- CD32 mode is giving an issue with Blue button. Tested with JoyPortTest and some WHDLoad games, seems that using the Blue button in CD32 mode eventually generates bursts of random combinations of other buttons. This may cause undesired effects in games (i.e. triggering pause or even quit combos). This happens only with Blue button, regardless of its mapping or if it is in auto fire mode or not. Other buttons work fine. With a standard CD32 gamepad the Blue button works correctly.

All tests were performed with a Logitech gamepad and a Sony DS4. The A1200 is an Escom version (should be Rev. 2B) with 8 Mb RAM expansion and the CD32 is equipped with a TerribleFire TF330 (68030 and 64 Mb of RAM). On the CD32, the issue is present without expansion and using a CD loaded game as well.

I hope this information is useful to help improving this wonderful device and I am available to perform some other tests.

If needed, I can produce a video of this issue.

Thank you for your support

willyvmm, 2022-01-06:
Sounds strange.
Can you confirm that the issue exists with both your mouSTer's ?

DigiTurk84, 2022-01-07:
Yes, I confirm the issue on both mouSTers.

willyvmm, 2022-01-26:
Could you tell me what software are you using to test this feature?
(I have only stock A1200 for tests)

DigiTurk84, 2022-01-26:
I am using JoyPortTest (http://aminet.net/package/driver/input/JoyPortTest.lha), but any WHDLoad game with CD32 gamepad support will expose the issue (i.e. Parasol Stars, Alfred Chicken).

As I said, this issue is present on A1200 with Rev. 2B motherboard and 8 Mb RAM expansion and CD32 both expanded and unexpanded with WHDLoad and CD loaded games (tested with original Alfred Chicken CD).

I can try to perform a test on A1200 without expansion.

TomCat666, 2022-03-17:
I can confirm this issue. Just got 5 mouSTers and this issue happens on all of them in cd32 mode. The blue button sometimes activates other buttons (mostly PLAY it seems) - tested with the both joy test programs on aminet and also with super frog where it goes into PAUSE mode randomly when pressing the Blue button. 
This is using the DualShock 3 joypads on Amiga 500.

DigiTurk84, 2022-05-01:
I performed a test on unexpanded A1200 with the same outcome.

However, totally by chance, I tested the controller using an interlaced resolution (both High and Super-High resolution interlaced), and the issue does not happen with JoyPortTest.
Did not test any game though.

willyvmm, 2023-01-25:
Please try this firmware: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3222

TomCat666, 2023-01-28:
Tried the latest firmware from a few hours ago and all the issues with cd32 emulation are gone. Working really nicely. Thank you.

willyvmm, 2023-02-06:
Thanks for Feedback.

closed 2023-02-06 by willyvmm