willyvmm/mouSTer #9: Mouse scroll wheel does not work in AmigaOS 3.2.1

lovrenco, 2022-01-10:
Just received mouSTer, uploaded firmware: mouSTer.fw.3.16.2600.zip

In Amiga OS 3.2.1 on Amiga 1200 middle mouse scroll wheel does not work, although it is switched on in the system. See attached photo. 

If you need me to test BETA app for this to resolve, feel free to contact me.

[image]

lovrenco, 2022-01-15:
Is there any info about this? Retrohax (Peter) told me to report issues here, I hope I am at right place?

csharpe, 2022-01-19:
I have the same behaviour on an Amiga 1200, 8 MB of RAM, 2MB Chip, OS 3.2.1 in ROM (not modules), mouster firmware 3.16.2600.  I have tried several mice and all have the same issue.  Only changes to the default config file was remapping mouse buttons so that the right mouse button is assigned to MB 2.

Is there anything you may suggest I try/check?

willyvmm, 2022-01-26:
@lovrenco Sorry for the delays I have Covid@home.

The Beta Firmware is fusked-up and is not working.

Afebriwyn, 2022-04-03:
Hello, is there any news on this topic? Thanks in advance :)

lovrenco, 2022-04-10:
Hello, I would like to test this fix
 When will be available, what do you think?

lovrenco, 2022-08-04:
Hello, any news about this fix?

SpawnTerror, 2022-08-07:
This project is dead now.

Afebriwyn, 2022-08-08:
> This project is dead now.

At least it looks quite dead - unfortunately...

Mangerio, 2022-12-29:
I have OS 3.2.1 on a A1230. Scroll wheel enabled. When i push down the scroll wheel it is recognized as a third mouse button. When i push down the scroll wheel and move the mouse up or down i can scroll the content of the IBrowse window. So i guess the Amiga scroll workaround is not implemented or activated in the mousSTer firmware?

ziggystar12, 2023-01-03:
No scroll wheel here either. And with this test firmware 1351 C64 support is broken as well. Thankfully I could downgrade firmware. It appears broken in 2600. I had to go back 4 versions.

willyvmm, 2023-01-23:
Please try the new pre-release 3.17.3150 build: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3150

Afebriwyn, 2023-01-24:
> Please try the new pre-release 3.17.3150 build: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3150

Tried it: works in conjunction with the micromys driver.
Out of curiosity: Will you provide your own compatible driver later, or is the plan to use the driver from iComp?
If there will be no own driver implementation: Couldn't there then theoretically be legal problems with iComp?

Mangerio, 2023-01-24:
Scrollwheel works well with the micromys.lha driver!
Thank you! ♥️ Amiga forever!

willyvmm, 2023-01-24:
> Out of curiosity: Will you provide your own compatible driver later, or is the plan to use the driver from iComp?
> If there will be no own driver implementation: Couldn't there then theoretically be legal problems with iComp?

I don't think we need another standard for Amiga mouse. 

Regarding legal issues: 
All work was performed with high ethical standards. Using an oscilloscope, Excel, and official website information.
No code was dissembled, I even don't have any micromys adapter, so I'm not 100% sure if my protocol implementation is 100% compatible. That will also explain the long time ... and initial issues.

willyvmm, 2023-01-24:
I would really appreciate it if someone would test it on a **PURE NTSC** machine.

ziggystar12, 2023-01-24:
I will test on a ntsc machine today. Also there is another method for scroll support using middle button and y. It’s supported out of the box on os 3.2.1

ziggystar12, 2023-01-25:
Working great on NTSC machine with no driver needed on 3.2.1! Now if you can fix 1531 also, we are rocking and rolling. :)

ziggystar12, 2023-01-25:
I did notice it seems a little sensitive, I think I can overcome that with a tool on Workbench though.

willyvmm, 2023-01-25:
The timeline is Amiga Wheel, CD32 Pad (almost fixed), 1351 (I can also emulate 1531 ... but I guess it's a misspelling :)

> I did notice it seems a little sensitive, I think I can overcome that with a tool on Workbench though.
On Both NTSC and PAL ? or only one of them?

> I will test on a ntsc machine today. Also there is another method for scroll support using middle button and y. It’s supported out of the box on os 3.2.1
So. Amiga os 3.2.1 has been released December 2021 - that is few months after I did a research around Wheel support. Basically I was not aware of that :)

Could You please point me to the technical details of this protocol?

Afebriwyn, 2023-01-25:
> Regarding legal issues: All work was performed with high ethical standards. Using an oscilloscope, Excel, and official website information. No code was dissembled, I even don't have any micromys [...]

About the legal problem: All right, was pure curiosity that moved me to this question.
On the other hand, with the current solution, it would be difficult for new users of Mouster if iComp would prevent the free download of the driver. 

I experimented a bit more with the firmware yesterday, for example I adjusted the sensitivity of the pointer several times. Suddenly, the mouse wheel no longer worked. After a power cycle, the problem was solved. I will continue to observe, perhaps I can trace a systematic...

ziggystar12, 2023-01-25:
I wasn't able to find any specific support articles on how Native Wheel is supported in OS 3.2, however it's been supported since 3.2, not necessary for 3.2.1.

willyvmm, 2023-01-25:
Seems like the "micromys" protocol has been adopted as the default wheel protocol. It is using exactly MiddleButton/Pot Y pin to transfer data,

ziggystar12, 2023-01-25:
Cool. When you are ready, I've got loads of old macs and apples with the 9-pin we can try to get MousTR working on those :) I realize you got other tasks. Thanks for getting this wheel to work. That makes it one of the only mutli-mouse-USB devices that can do Amiga Wheel right now.

Afebriwyn, 2023-01-30:
> Working great on NTSC machine with no driver needed on 3.2.1! Now if you can fix 1531 also, we are rocking and rolling. :)

Do I understand you correctly?
You use the Mouster adapter and a USB mouse, the scroll wheel works for you out-of-the-box without any additional tool or driver?
For me (A500-PAL, OS3.2.1, option enabled in Prefs) the scroll wheel only works when the Micromys driver is running in the background.

ziggystar12, 2023-01-30:
Upon closer inspection, I thought Micromys was a Startup folder option. I did actually have it installed in s:user-startup   So you are correct, it does require Micromys to work.

DownshiftDX, 2023-01-31:
Is there anything else specific to getting the Micromys driver to work? I have a pretty heavily expanded A500 (NTSC) with a Vampire V2 and with Micromys in user-startup it seems to initially corrupt the pointer until I click, Magic Menu doesn't load, and I have no scroll wheel even though it is enabled in IControl. I just got done confirming my mouse was detected properly with the latest pre-release build. (3.17.3377)

DownshiftDX, 2023-01-31:
> For kicks try the firmware two versions ago that initially fixed the wheel. I’m on a pistorm and I haven’t checked wheel on 3377 yet. Get Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

Going back to 3.17.3150 I was able to get scrolling to work with my Logitech mouse but not my cheap ONN one that was just added in the latest build.

ziggystar12, 2023-01-31:
Ahh yeah, I've had almost 0 luck getting the cheap mice to work. Logitech seem to work great though.

DownshiftDX, 2023-01-31:
I may go buy another Logitech tomorrow but I am pretty thrilled that willyvmm was able to take my device data and make the ONN work so quickly. I am sure it is just a regression with the 3377 build.

DownshiftDX, 2023-02-03:
> Further feedback about Mouster under Amiga OS 3.2.1: Something strange is going on with the adapter or the Micromys driver, at least on my Amiga. I noticed it with a game which does not accept joystick inputs when the Micromys driver is running in the background. I installed a joystick test program for the workbench (http://aminet.net/package/driver/input/JoyPortTest). If this is started with the Micromys driver activated, it looks to me as if the test program is flooded with input events, so that the GUI of the test program is no longer operable. Sometimes it also leads to a crash of the computer. Without the driver everything is normal, the joystick inputs are displayed as expected in the test program. Other joystick test programs behave similarly.

I found just now that the Micromys driver seems to break when a joystick testing program is loaded. I used Psygore JoyPortTest v0.1 I didn't see it flooded with input events nor have I had a crash, just noticed the scroll wheel wasn't registering on the Y button like I expected and then figured out that scrolling had stopped working entirely. I'd been trying to figure out why I was having problems with scrolling working very inconsistently after a cold boot and MagicMenu being disabled after it had started. (If scrolling worked, MagicMenu was not or the other way around.) After a few different tests with failsafe boot and disabling things in startup-sequence it seems like maybe StackAttack starting before Micromys was possibly the issue. (I have re-arranged my startup several times now.) I don't think it is a problem with the mouSTer though... I think it is probably something buggy about Micromys driver.

A500 with Vampire V2+ core 2.16 w/OS 3.2.1

Afebriwyn, 2023-02-16:
At least for me, the problem with Psygore's tester has been resolved by installing a different, experimental driver:
https://forum.icomp.de/index.php?thread/2175-micromys-and-os3-2-mouse-support/