willyvmm/mouSTer #10: iiglo M310 mouse not working (Atari ST)

evldhs, 2022-01-15:
iiglo Travel Mouse
Model number: M310
P/N: IIM310WL
Barcode: 7070205140330

Tested with  Atari STe and Atari Falcon
Mouster firmware 3.16.2600

The wireless mouse (with USB dongle) does not work. Mouse pointer does not move in any direction and no clicks seem to work. 
The mouse works fine with a PC or Mac and also with the Smallymouse2 adapter for the Atari.

Other mice work fine with the same Mouster.

Configuration is set to force mouse and Atari mode.

See attached zip for debug file and ini file.
[mousterdebug.zip](https://github.com/willyvmm/mouSTer/files/7874891/mousterdebug.zip)

Copy/paste of the debug file:

3.16.2600
1EA7
0064
06 B5 FF 09 01 A1 01 85 B5 09 02 15 00 26 FF 00
75 08 95 07 81 02 09 02 15 00 26 FF 00 75 08 95
07 91 02 C0 05 01 09 02 A1 01 85 02 09 01 A1 00
05 09 19 01 29 08 15 00 25 01 95 08 75 01 81 02
05 01 09 30 09 31 16 01 F8 26 FF 07 75 0C 95 02
81 06 09 38 15 81 25 7F 75 08 95 01 81 06 05 0C
0A 38 02 95 01 81 06 C0 C0 

Copy/paste of the ini file:

;
;
; mouSTer - universal USB HID class device to db9 adapter. 
; Firmware revision: 3.16.2600

;
;                                          _/_/_/  _/_/_/_/_/                  
;    _/_/_/  _/_/      _/_/    _/    _/  _/            _/      _/_/    _/  _/_/
;   _/    _/    _/  _/    _/  _/    _/    _/_/        _/    _/_/_/_/  _/_/     
;  _/    _/    _/  _/    _/  _/    _/        _/      _/    _/        _/        
; _/    _/    _/    _/_/      _/_/_/  _/_/_/        _/      _/_/_/  _/         
;
;
; ###############################################################################
; #                                                                             #
; # Features:                                                                   #
; # Emulates Atari, Amiga and Commodore 1351 mouse.                             #
; # Can handle up to 16 buttons that can be translated to an action*            #
; # Can handle up to 6 axes, that can be translated to an action*               #
; #                                                                             #
; # More to come...                                                             #
; #         *an action - means any joystick direction or fire/mouse button      #
; #                                                                             #
; ###############################################################################
;
; ###############################################################################
; # Users Manual                                                                #
; ###############################################################################
;
; ###############################################################################
; # WARNING: WARRANTY MAY BE VOID IF USED WITH Atari 7800                       #
; ###############################################################################
; # Atari 7800 IS NOT SUPPORTED:                                                #
; # Please do not use the mouSTer with Atari 7800.                              #
; # It may be dangerous for the mouSTer due to weird A7800 construction.        #
; ###############################################################################
;
[... 380 lines cut ...]
; [08|paddleB|potY] - axis is directly mapped to analog paddle B. requires SID.
; [09|lb|fire] - axis is one direction fire button.
; [10|rb|fire] - axis is one direction button mapped to paddle A.
; [11|mb|fire] - axis is one direction button mapped to paddle B.

axis01=ns
axis02=ew
axis03=none
axis04=none
axis05=none
axis06=none

; validate=
; Validate if pressed direction button are giving valid joystick direction
; valid states are: n, ne, e, se, s, sw, w, nw or neutral
; invalid states are fx: ns, ew, new ,nesw etc.. When invalid state is detected, the neutral state is applied
; buttons nor paddle never are validated.
; Default:
; validate=false
;
; Possible values:
; [00|false|off] - direction states are not validated.
; [01|true|on] - direction states are validated.

validate=false

willyvmm, 2022-01-26:
Thank You for the debug data.

willyvmm, 2023-01-28:
Please try the latest Pre-Release Firmware: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3309

evldhs, 2023-01-28:
Thanks for the update.
I'm afraid that the iiglo M310 is still not recognized by the Mouster after upgrading to beta firmware 3.17.3309.
Other wireless mouses work fine (for example Logitech M190) with the same Mouster.

The iiglo M310 works fine on a PC and Mac.

willyvmm, 2023-01-29:
Ok. I think I've found the root cause. 
Please try this: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3314

evldhs, 2023-01-29:
Hi,

thanks for the update again and yes this solved the problem!
iiglo M310 mice now works with Mouster, thanks a lot!

evldhs, 2023-01-29:
Firmware 3.17.3314 resolved the issue.

closed 2023-01-29 by evldhs