willyvmm/mouSTer #11: Pre-Release - wheel testing (C64)

kbx0, 2022-01-17:
I have an Ultimate64 Elite with 2 SIDs.

I have tested the latest mouster update with 2 wireless mice and 1 wireless mouse in mousetester2.

Sorry to say that middle click and wheel do not work at all.

I also have the MicroMYS, with the wired ps/2 to usb mouse, the middle click and wheel works.

Need testers?

willyvmm, 2022-01-26:
#9

gnacu, 2022-12-30:
I have just completed Wheel support at a low-level in the C64 OS KERNAL and UI Toolkit, with a driver for MicroMys and updates to the Mouse settings Utility to adjust direction and sensitivity. If you guys release support for mouse wheel for MouSTer, I promise I'll write a MouSTer driver for C64 OS immediately. Mouse Wheel support rolls out for C64 OS very soon, in the next update, v1.02.

willyvmm, 2023-01-30:
Please try the latest firmware: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3377

markusC64, 2023-01-30:
I confirm the lastest firmware 3.17.3377 works on Ultimate 64 with wired mouse. Full Micromys functionality verified - incl. all 3 buttons and wheel.

willyvmm, 2023-02-06:
As the C1351 have been rebuild once again, please try the latest build:  https://github.com/willyvmm/mouSTer/releases/tag/3.17.3455

gnacu, 2023-03-04:
Sorry for the late response, I just tested this latest firmware, and it works perfectly, as far as I can tell. I tested the mouse wheel in C64 OS and it's working great.

Greg.

kbx0, 2023-05-11:
Sorry for the delay in responding.

Tested:  Logitech M525 Wireless - Razer Deathadder V2 Pro Wireless

Machine: Ultimate64 Elite on 3.10c with Dual Arm2Sid.

Software: Joyride 1.6

Results: Three working buttons and a working scroll wheel.

Thank you so much, you've made the Mouster so useful.

closed 2023-05-11 by kbx0