willyvmm/mouSTer #13: Onn. Wireless Mouse not working (Amiga)

DownshiftDX, 2022-02-05:
Wireless ONN branded mouse from Walmart that I use on my main Amiga doesn't work with mouSTer on fw 3.16.2600.  This wireless mouse does work with the Tom+ adapter I got a while back before I knew of  the mouSTer. 
The mouSTer shows the heartbeat like it is functioning properly but the mouse is unable to move.

Tested on two different A500s on different PSUs where all other mice work. The mouSTer does work with controllers and other wired mice I have tried.

Debug attached.
[MOUSTER.DBG.TXT](https://github.com/willyvmm/mouSTer/files/8008820/MOUSTER.DBG.TXT)

The mouse label has the following information:
ONN. Wireless Mouse
Stock No: 100012634
FCC ID:PRDMU79

willyvmm, 2023-01-30:
Please try the latest firmware: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3377

DownshiftDX, 2023-01-31:
ONN mouse is now detected and works.

closed 2023-01-31 by DownshiftDX