willyvmm/mouSTer #14: Feature Query: Megadrive / SMS support plans?

Xalior, 2022-02-11:
Do you have any plans to add MegaDrive / Sega Master System support?  I'm keen to try this on the Spectrum Next, which has native support for this controller type (as an extension of Kempston, logically enough) and wondered if it was on your radar?

willyvmm, 2022-12-16:
MD has another pinout, so it's not straight forward. It requires modification on the PCB layout. 
Answering your question: YES it is planned but i need more feedback if there is someone who would like to buy it.

Echtze1t, 2024-01-08:
> MD has another pinout, so it's not straight forward. It requires modification on the PCB layout. Answering your question: YES it is planned but i need more feedback if there is someone who would like to buy it.

Hi, I would like to help you in any way possible to get the Mouster working with the Sega Mega Drive. Please let me know what I can do to help.

smf-, 2024-01-12:
I got a spectrum next, so I too would like MD support. I assume the PCB modification would require to buy a new one?

willyvmm, 2024-01-28:
> I got a spectrum next, so I too would like MD support. I assume the PCB modification would require to buy a new one?

Yes. SEGA pinout is different. 5v and GND are in different places.
An alternative may be a nasty Adapter.