willyvmm/mouSTer #15: NOMAD GTX-560 TRUST

DiabloDesign, 2022-02-24:
My PAD does not work NOMAD GTX-560 TRUST I am sending Mouster.dbg

3.16.2600
145F
0232
05 01 09 05 A1 01 15 00 25 01 35 00 45 01 75 01
95 0D 05 09 19 01 29 0D 81 02 95 03 81 01 05 01
25 07 46 3B 01 75 04 95 01 65 14 09 39 81 42 65
00 95 01 81 01 26 FF 00 46 FF 00 09 30 09 31 09
32 09 35 75 08 95 04 81 02 06 00 FF 09 20 09 21
09 22 09 23 09 24 09 25 09 26 09 27 09 28 09 29
09 2A 09 2B 95 0C 81 02 0A 21 26 95 08 B1 02 0A
21 26 91 02 26 FF 03 46 FF 03 09 2C 09 2D 09 2E
09 2F 75 10 95 04 81 02 C0

willyvmm, 2023-01-28:
Please try the latest Pre-Release Firmware: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3309

DiabloDesign, 2023-02-01:
He's not cooperating

willyvmm, 2023-02-06:
Try the latest revision here: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3455
I tested the debug data under my emulator and ... should work.