willyvmm/mouSTer #16: Mac Plus and Apple II compatibility

supercompman, 2022-04-04:
It would be great to have compatibility with the Mac 128k/512k/Plus mouse which can also be made to be compatible with the Apple II mouse. Look [here](https://blog.rekawek.eu/2016/12/08/mac-plus/) for information about how to use an Amiga mouse on a Mac Plus. Look [here](https://www.raphnet.net/electronique/apple_iic_mouse_fixer/index_en.php) for information about making a Mac 9-pin mouse compatible with the Apple II series.

willyvmm, 2022-12-16:
As well as MD Mac is also planned for the future. 
I have to analyze signal requirements for the M0100 mouse and take a decision if it is possible by firmware mod only or it needs a hardware adapter.

ziggystar12, 2023-01-31:
Amiga mouse adapter         ST mouse adapter

Mac Plus  Amiga mouse     
9W male   9W male         
1--GND------8          
2--VCC------7             
4--XA-------2             
5--XB-------4              
7--Button---6+9           
8--YB-------3             
9--YA-------1

ziggystar12, 2023-01-31:
Pinout for Mac Mouse from Amiga
Swap 1 and 8 for GND
Swap 2 and 7 for VCC
Swap 4 and 2 for XA
Swap 5 and 5 for XB
Combine 6+9 on Amiga to 7 for button  (unsure if either 6 or 9 will work, I'm guessing yes because Mac is 1 button)
Swap 3 and 8 for YB
Swap 9 and 1 for YA

ziggystar12, 2023-01-31:
Looks like APPLE II is much more complex.

willyvmm, 2024-01-28:
I was thinking about APPLE II recently ... and I think there may be a better solution.
What do you think about a mouSTer USB mouse card for APPLE II ?

ziggystar12, 2024-02-17:
https://www.applefritter.com/content/a2usb-apple-ii-usb-mouse-interface-card-emulation

 

https://retroconnector.com/products/desktop/adapters-and-interfaces/usb-mouse-interface-apple-ii/

 

Some things out there. Wish we could just add a little dongle between the existing adapter and the apple ii instead. 😊

 

John

ziggystar12, 2024-02-21:
I could see the two angles. Obviously from a cost perspective and to now say you have a single device that can handle more old computers than any other the dongle. Another approach in a future device might be to have a physical switch in the same device to switch those lines.

A dedicated device sounds like a cost/development risk to you and you also begin to compete with a dedicated product in a space that has some already.

There is no single mouse USB device that does what we are proposing here today. So that's marketable and a differentiator.

That is also possible.
That solution would extend the overall length of the mouster by ca. 15-30 mm:
1708551019677.jpg (view on web)<https://github.com/willyvmm/mouSTer/assets/24528286/f138ac8e-2565-470d-ab34-0a8436488795>

OR, a dedicated mouSTer for Mac.
For my curiosity, what solution would be best for You?