willyvmm/mouSTer #17: 1351 mode glitches

TomCat666, 2022-04-08:
Hi
Using latest version on c64 with 1351 mode turned on and using any multiplier I get occasional glitches in both directions. For 1 frame the mouse (or bat in arkanoid) would jump (delta would get some odd value) then in next frame it would jump back to the previous position (delta would get the negative same value). I tried with 3 monsters and it is the same.

willyvmm, 2023-01-18:
I will lok into later. New driver for analogue mouse is on its way.

willyvmm, 2023-01-30:
Please try the latest firmware: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3377

ziggystar12, 2023-01-30:
Testing now.

ziggystar12, 2023-01-30:
Works!!

ziggystar12, 2023-01-30:
Does not actually work correctly on GEOS, after further testing it appears it is moving to fast for GEOS. I thought it moved pretty quickly on Beholder and thought, this was nice! However, it makes the mouse jump all around the screen on GEOS unfortunately. So I think you just need to make the sensitivity a little less.

ziggystar12, 2023-01-30:
Could be - but I noticed the mouse is a much faster on EOB then it was before. Even without Acel option.

ziggystar12, 2023-01-30:
Also, firmware 2019 C1351 worked great in both EOB and GEOS

willyvmm, 2023-01-30:
Try this Firmware:  https://we.tl/t-NqNx1OknEs mouse is now slowed down.
Also, the parameter in config (microstep=02) may be adjusted.

ziggystar12, 2023-01-30:
This unfortunately made things worse for GEOS. Even with no dongle/mouse connected, the mouse jumps around the screen randomly.

TomCat666, 2023-01-31:
Hello. Thank you for this update!
I have tested it now on my Ultimate64 and (almost) everything works:
Joyride, EoTB, Supremacy (Excess), Geos (1.2, TopDesk, Wheels), Supremacy, Krakout 1351, Deja Vu (n0stalgia) all work fine
The only thing I couldnt get to work is left mouse button (fire) in Cabal (n0stalgia version)

Working fine with the wired USB mouse (even mouse wheel works in Joyride :) )

When I have connected my logitech wireless usb mouse (M187 mini mouse) the keyboard completely stops working - it registers keypresses all the time, even if I don't move the mouse at all.

ziggystar12, 2023-01-31:
> Hello. Thank you for this update! I have tested it now on my Ultimate64 and everything works: Joyride, EoTB, Supremacy (Excess), Geos (1.2, TopDesk, Wheels), Krakout 1351, Deja Vu (n0stalgia) all work fine. I had to slow it down a bit with microstep=03 otherwise it was jumping too much in some games - the overflow problem. Seems microstep 03 is good for me in the amiga mode too.
> 
> GREAT work ! :) Both cd32 and 1351 mouse working as they should.

What happens if you remove the usb mouse dongle on GEOS? Can you try GEOS 2.0?   Also, here is a mouse driver to enable either port. https://commodore.software/downloads/download/662-geos-input-device-drivers/15900-port-2-drivers-v1-1-geos

TomCat666, 2023-01-31:
> What happens if you remove the usb mouse dongle on GEOS? Can you try GEOS 2.0? Also, here is a mouse driver to enable either port. https://commodore.software/downloads/download/662-geos-input-device-drivers/15900-port-2-drivers-v1-1-geos

I couldn't get these drivers to work (for some reason they don't even show up in the folder or input selector), but built-in 1351 driver of 2.0 works fine with mouster latest fw...

willyvmm, 2023-02-06:
The C1351 engine got merged all the best from 2600 build and the "new engine".
Please try the latest build:  https://github.com/willyvmm/mouSTer/releases/tag/3.17.3455

ziggystar12, 2023-02-07:
So far so good on 1351!

Retro-Nerd, 2023-02-07:
Yep,  the C351works fine now. Tested all functions, via Joyride tool. Good job!

vandabbin, 2023-03-02:
working great for me with most mice but when I use the a500mini themouse it will occasionaly freeze the pointer in place while the mouse is still moving in a direction and requires me to either lift the mouse up and put it down to get it to work again.

the mouse works fine with every other device if used it on

willyvmm, 2023-03-02:
@vandabbin Seems like it's not related to the mouSTer - rather to the connected device.
A problem with the mouse has been reported to me before and exists only with "the mouse". I'm almost sure that the cause of the problem is the not fully compliant implementation of the USB STACK in the device. 
I know I push the USB HID interface to the limits (but not over the limits) to minimize latency. And this is the only device I know that is causing problems.

JanNadziejowski, 2023-03-02:
Hello, The situation is like this. I uploaded FW 3.17.3475004. I checked the config, FW version and type (amigamm). I launched WB3.1 on RTG on Pistorm 030. Of course FreeWheel and micromys fired. Unfortunately, the roller does not work. I turned off the mouster, plugged it in again and downloaded the config. The config is 3.8.1743 FW. What can be played?

vandabbin, 2023-03-02:
Ah I see, thanks for the info. Glad it's not the mouster then, too bad about 'the mouse' not working though it matches so nicely haha!

willyvmm, 2023-03-02:
@JanNadziejowski 
That's the point. The update process is too complicated and too restricted.
Remove everything from the mouSTer folder, and put the Firmware Only.

Btw. Wrong thread.

JanNadziejowski, 2023-03-02:
Ok. Willy, sorry. I'll try again and post in the right thread haha  :P

closed 2023-11-06 by willyvmm