willyvmm/mouSTer #20: moister firmware updating to latest available

martingclark, 2022-06-23:
tried to update this evening to the latest firmware available, downloaded onto 2 different usb sticks, plug in the mouster, insert the usb stick but all I'm getting is 2 flashes and then one more a bit further apart it writes to the usb and makes a folder called mouster with an ini file in it but then when I open the ini file confirms its still on the older firmware 3.12.2019 and not 3.16.2600 that I've downloaded, Any ideas why.

Thanks

creopard, 2022-06-24:
Did you put the firmware file in the root directory of the USB-stick or in a subdirectory called "MOUSTER"?
I think that makes a difference.
Besides that, the stick must not contain a MOUSTER.INI file, otherwise the firmware is being ignored as it would just update the MOUSTER.INI file.

martingclark, 2022-06-24:
Root

⁣Get BlueMail for Android ​

willyvmm, 2022-12-16:
To update firmware please remove MOUSTER.INI from the MOUSTER folder and put the new firmware file only.

closed 2022-12-16 by willyvmm