willyvmm/mouSTer #22: Project Dead? @willyvmm?

thesnable, 2022-08-30:
@willyvmm 

Is this Project dead? I got my mouster from retrohax yesterday, and the usb competition pro and the joystick from thec64 isnt working unfortunately.

I thought, you wanted to keep this updated?

Why dont release the source then?

Afebriwyn, 2022-10-11:
If there is no more interest to continue the project, it would be a good idea to release it as open source in the hope that someone else will maintain and improve it.
I actually bought the mouster only because there was already a beta firmware with mousewheel support. Otherwise I would have bought the Micromys right away...

willyvmm, 2022-12-16:
[https://github.com/willyvmm/mouSTer/issues/23#issue-1500949553 #](https://github.com/willyvmm/mouSTer/issues/23#issue-1500949553)

closed 2022-12-16 by willyvmm