willyvmm/mouSTer #23: Hard time.

willyvmm, 2022-12-16:
Hi everyone. 

To have it quick.
I have had hard time last months, but seems like everything is on good path.
I was continuously working on the project, but with very limited amount of time. VERY VERY limited. 
I read all messages on my email as well as github updates  etc. on a regular basis, but had no motivation to write any reply.
You may call it Depression or Procrastination or ... whatever You want, but i never drop the project.
So. 
I hope i would be able to release new Firmware before X-mass.
Also, as some of you have suggested, i´m considering to release something like an SDK!
So, you would be able to develop your own piece of software based on the built-in bootloader.

Best. Willy.

Afebriwyn, 2022-12-16:
Hi Willy,
thank you very much for your reply!
I think I speak for most when I say that our main concern was the lack of communication. We believe in the product itself, and if the development continues after a year, I for one am happy and grateful. Also great is the info that there will eventually be an SDK - something we can all certainly benefit from!

thesnable, 2022-12-17:
@willyvmm All the best from me. If you need help, dont hesitate to contact me. I (we) really appreciate your Project, and maybe use it to have a little sun on the horizon.

Merry 🎄

TomCat666, 2022-12-17:
Hi Willy.

Thanx for the heads up that the project is still in progress. I wish you get better soon, health is much more important than anything really! Take care and have a nice xmas!

lovrenco, 2022-12-18:
Nice to hear from you Willy.
Health/private life comes first, of course. Take care, we can wait.
Merry Christmass and happy New year!

ziggystar12, 2023-01-03:
This really has the potential to be the best USB/DB9 Mouse adapter there is. Just a few kinks to work out. Glad you are doing better. Happy new year!

Jumpman64, 2023-01-18:
Good to hear from you that you're ok!
mouSTer is working fine already and it's always more important, that you're doing fine too.
Let's see what 2023 will bring. mouSTer rocks!

willyvmm, 2023-01-18:
So. The good news is I have identified the root cause.
Loooong time ago I made a completely wrong assumption. As a result, I lost dozens of hours trying to hunt a bug that did not exist. Finally the bug is identified, and I'm working on a completely new version of the wheel and 1351 driver.

Please be patient, the new firmware is coming.

closed 2023-01-23 by willyvmm