willyvmm/mouSTer #24: Mouster cd32 joypad emulation doesnt work correctly in some games

moulder625, 2022-12-29:
Hi,

I have dual shock 4 connected to Mouster with cd32 joypad emulation on. In some games adapter is't working correctly. For example in Flashback unnoficial cd32 port joypad is clicking buttons on its own when splashscreen with language and difficulty selection is shown. Also in Skeleton Crew after few seconds game exits to main menu because mouster is emulating back button click on its own. Are you planning to release new firmware with bug fixes?

willyvmm, 2023-01-18:
#8

willyvmm, 2023-01-25:
Please try this firmware: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3222

moulder625, 2023-01-25:
Ok,.Will do.that.

Thanks.and.regards,
Tomasz

TomCat666, 2023-01-28:
Working fine for me, tried several games that support cd32 pad and workbench tools that display cd32 button activity. No more ghost (or wrong) button presses. Thanx!

willyvmm, 2023-02-06:
Thanks for Feedback.

closed 2023-02-06 by willyvmm

moulder625, 2023-03-25:
Ok, so I've tried newest beta firmware and there is progress for example Skeleton Crew is working correctly without wrongly pressed buttons.
 But Flashback unnoficial port for CD32 with cd32 joypad support is not working correctly - mouster is pressing buttons on itself on few splashscreens where there are options to choose start level, difficulty, language and so on.