willyvmm/mouSTer #25: PS5/DS5 controller not working.

sho3string, 2023-01-26:
Retrohax lists Sony DS5(wired) as supported but the device does not detect the controller at all when plugged in. Tried all available firmware.

DS4/PS4 controller working  fine.

willyvmm, 2023-01-26:
I'm not sure why, but I can confirm it's no longer recognized by mouSTer. I guess that may be caused by DS5 firmware update?

sho3string, 2023-01-27:
That's likely the case.

willyvmm, 2023-01-27:
Could you please try the latest prerelease firmware form yesterday: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3291
I'm not 100sure how the DS5 is working (Remember that it is a dedicated controller). Yesterday/2 days ago I could not get the connection. Today it got connected ... without any additional action.
To be honest - I don't understand why?

sho3string, 2023-01-28:
Unfortunately it's not working for me. I tried another brand new controller I had un-opened since I bought my PS5, then updated firmware on controller using 'Firmware updater for DualSense", updated mouSter firmware.  Tried changing  the mode in the INI.

willyvmm, 2023-01-28:
OK. I can see some differences. I have performed a HARD RESET (small hole on the back https://www.online-tech-tips.com/wp-content/uploads/2022/11/how-to-reset-your-ps5-dualsense-controller-6-compressed.jpg ) on my DS5 controller. Seems like i have to update the firmware once again and try again.

willyvmm, 2023-01-28:
OK. I can see some differences. I have performed a HARD RESET (small hole on the back https://www.online-tech-tips.com/wp-content/uploads/2022/11/how-to-reset-your-ps5-dualsense-controller-6-compressed.jpg ) on my DS5 controller. 
I have verified the DS firmware and the tools say 0307 (the latest).

Could you try with the latest firmware: https://github.com/willyvmm/mouSTer/releases/tag/3.17.3309

If still not working, try to rest using the hole on the back.

willyvmm, 2023-11-06:
The workaround until I understand the root cause of this issue is to hard reset the DS5 controller.

closed 2023-11-06 by willyvmm