willyvmm/mouSTer #26: 8bitdo Controller Support (for CD32 pad emualation)

Retro-Nerd, 2023-01-28:
Hi,

any chance that you could add CD32 pad emulation for the 8bitdo Pro+/Pro 2 controllers wired and wireless (prefered). They are practically Xbox 360 protocol type controllers. I tried it with your latest beta firmware which has CD32 pad emulation as default enabled. It kinda works, but only the directions and Amiga button 1.  Also tried to change the 01-04 game pad numbers, but no luck.

Retro-Nerd, 2023-01-28:
Argh!  Looks like my used USB stick isn't properly detected at times.  Didn't check if the new file was really uploaded to the adapter. Anyway, the update worked for me. . I can confirm that beta firmware 3.17.3291 behaves fine with all CD32 and WHDLoad CD32 controller patched games on my A1200. Tested the 8bitdo Pro+ with my 8bit Dongle. No issues. Also tested the Pro 2 wired (X-Input mode) and wireless (Android mode). Also no problems at all. Thanks for making this possible!

willyvmm, 2023-02-06:
Thanks for Feedback.

closed 2023-02-06 by willyvmm