willyvmm/mouSTer #28: Amiga mouse scroll wheel still not working in latest.

lipteklips2k15, 2023-02-08:
cannot get amiga mouse scroll wheel to work in any of the firmwares that are supposed to support it including the latest! Mouse works just no scroll! and i have the micromys driver working fine as using a usb mouse on my clockport usb interface scrolls fine.

willyvmm, 2023-02-08:
I'm not an Amiga guru, but it works for many people. 
I'm using a stock A1200 with only 2 drivers installed: freewheel and micromys.
Another issue may be the conflict of the "USB clockport" solution with the legacy attempt and the driver fx. prefers USB over legacy. 
Can you tell me what is Your Hardware and configuration, I will try to get support from some Amiga gurus (if they currently are not meditating).

Afebriwyn, 2023-02-09:
2 INI parameters that must be set appropriately - otherwise it didn't work, at least for me:
mode=mouse
type=amigamm
Maybe this already explains your problem?

lipteklips2k15, 2023-02-09:
> 2 INI parameters that must be set appropriately - otherwise it didn't work, at least for me: mode=mouse type=amigamm Maybe this already explains your problem?

those options are configured and still dosnt work.

Retro-Nerd, 2023-02-09:
I'm using firmware 3.17.3150 and the scroll is working fine. A1200, ClassicWB Lite, Freewheel, micromys driver installed. The later revisions didn't work for me.

lipteklips2k15, 2023-02-09:
> I'm using firmware 3.17.3150 and the scroll wheel is working fine. A1200, ClassicWB Lite, Freewheel and micromys driver installed. The later revisions didn't work for me.

just tried this firmware and still isnt working.

radius75, 2023-02-10:
Maybe check if the configuration is loaded.
Connect the USB flash drive with the empty MOUSTER folder and check what is written in the resulting INI file (version, config)

lipteklips2k15, 2023-02-11:
> 

already done that too.

Afebriwyn, 2023-02-16:
Worth a try: 
Experimental micromys-driver from here:
https://forum.icomp.de/index.php?thread/2175-micromys-and-os3-2-mouse-support/

willyvmm, 2023-02-16:
Could you please tell us your hardware configuration?
Could You please tell us your software (mainly drivers) configuration?
Have you tried to disable/remove the USB Driver?

midwan, 2023-02-16:
I'm afraid the mouse wheel didn't work for me either, in 4 different mice models I've tried so far.
Meanwhile, on the same hardware/software installation, using the Micromys adapter and a PS2 mouse, the wheel works as expected.

- Firmware revision: 3.17.3475
- tried with `mode=auto` and `mode=mouse` (no difference)
- set `type=amigamm`
- tried with the Micromys 1.1 experimental version as well (no difference)

This is on AmigaOS 3.2.1, which has native support for Wheel (when enabled), so the FreeWheel program is no longer needed. Only Micromys running should be enough (and it is, with the micromys adapter).

The mice models I tested were:
- Raspberry Pi mouse
- ASUS laptop model mouse
- HP desktop
- A cheap gaming mouse

The hardware I tested the Mouster on was:
- Amiga 4000/060 with AmigaOS 3.2.1
- Vampire V4 with AmigaOS 3.2.1

Happy to provide more details if you need them

Retro-Nerd, 2023-02-17:
Only for the record. I am using Kickstart/OS 3.1 on my A1200.  Firmware 3.17.3150 works fine for me (typical PC HID  mouse, wireless), but every revision after that not. So, it can't be an exclusive OS 3.2.X issue here. I followed this advice, and 3.17.3150 does the trick for now.

https://github.com/willyvmm/mouSTer/issues/9#issuecomment-1409662646

midwan, 2023-02-17:
I forgot to mention that I also tested 3.17.3150, so I updated my report above to indicate that. It didn't work for me either :(

willyvmm, 2023-02-17:
Hi.
Could you also tell me if the machine is PAL or NTSC ?

midwan, 2023-02-17:
All my machines are PAL

willyvmm, 2023-02-17:
Looks like kind of hardware incompatibility that i'm not aware of.
Is there a chance that you are a lucky owner of an oscilloscope?

midwan, 2023-02-17:
@willyvmm Unfortunately I don't have an oscilloscope :(

willyvmm, 2023-02-17:
@midwan can you remember when have you bought the mouSTer? 
Is there a chance that it's a mouSTer from the first pilot batch?

midwan, 2023-02-17:
@willyvmm 
From checking my mails: [Order #11977] (January 27, 2021)

myzar, 2023-02-20:
Same situation here order:
[Order #15150] (May 15, 2022)

A1200 PAL Machine , tried everything two different logitech mouse , no joy

willyvmm, 2023-02-21:
@myzar, @lipteklips2k15: please be patient, we're investigating this issue.
In the meantime, please inform me when you have bought the mouSTer.

lipteklips2k15, 2023-02-22:
a very long time ago - I cant be sure because a lot has happend since but at least a year.

gklka, 2023-02-22:
Hello! I've also tried the 3.17.3150 firmware today with the required settings (`mode=mouse` and `type=amigamm`). Installed Micromys driver from http://wiki.icomp.de/wiki/Micromys#Download. I have an A1200 equipped with Vampire V2 and a vanilla AmigaOS 3.2.1. I know, that in theory 3.2 support scrolling without the need of FreeWheel, but also tried to install that program, but unfortunately did not change anything. Scrolling is not working for me.

Tested mouses:
- Logitech Performance MX
- Logitech M280

MBry0, 2023-02-25:
If of any use, i tried 4 mice and they all work, scrollwheel included.

- mouster fw 3.17.3475
- Amiga os 3.2.1 (rom and wb) installed upon classicwb3.9 (then 3.1.4 -> 3.2 -> 3.2.1)
- no freewheel
- micromys drivers (the stable ones)

Mouse tested:
- generic Microsoft (HID\VID_045E&PID_0797&REV_0200)
- Logitech M100 (HID\VID_046D&PID_C077&REV_7200)
- Maxam SB-1000 (HID\VID_0000&PID_3825&REV_0100)
- Microsoft comfort mouse 4500 (HID\VID_045E&PID_076C&REV_0083&Col02)*

* this mouse didn't work at all on firmware 3.16.2600

myzar, 2023-02-25:
I think i've solved the mistery after trying almost everything, it's not an hardware/firmware problem , this microsys driver doesn't work on rtg , if i switch to a pal mode , scrolling works , as soon as i switch to a rtg screen scrolling stops working.

lipteklips2k15, 2023-02-25:
> I think i've solved the mistery after trying almost everything, it's not an hardware/firmware problem , this microsys driver doesn't work on rtg , if i switch to a pal mode , scrolling works , as soon as i switch to a rtg screen scrolling stops working.

My actual micromys mouse adapter works fine with rtg so its not that

myzar, 2023-02-25:
> My actual micromys mouse adapter works fine with rtg so its not that

well do a quick test switch to a pal screen and you'll see that magically it works

gklka, 2023-02-25:
I did test this on PAL screens, even with pure 3.1 Kickstart + loadwb only, but it does not work for me.

TuKo1982, 2023-02-26:
Tried here with 3475 and indeed it does work in PAL but not in RTG.

My configuration : 
- AmigaOS 3.9
- Picasso96 2.0
- PiStorm
- Micromys + FreeWheel

midwan, 2023-02-28:
Just got the (latest) adapter back, so I could test as well. Unfortunately, it still doesn't work for me, not even in PAL mode as mentioned above.

- AmigaOS 3.2.1
- P96 3.3.2
- Micromys driver 1.0 (FreeWheel is _not_ required in AmigaOS 3.2.x as it has native support for it)
- ZZ9000 RTG

I tried it under Workbench, DOpus and IBrowse, all of which work fine with scrolling when using an actual Micromys adapter and a PS2 mouse connected. RTG or PAL mode did not seem to make a difference in my environment.

lipteklips2k15, 2023-03-01:
still not working here across 5 different amigas some with RTG some just with PAL tried all firmwares that are supposed to support scrollwheel on amiga using different micromys drivers and freewheel and nothing! please fix this its driving me nuts :P

gklka, 2023-03-01:
What can we help you to locate the issue? Are there any debug tools or methods to identify the problem? (Apart from owning an oscilloscope...)

willyvmm, 2023-03-01:
So... I'm trying to obtain a Pistorm for A500 - as seems to be the only option to get RTG. 
That could be an advantage if someone have tested it with the Pistorm.
And ... oscilloscope is still the best tool. 
Until I understand what is cause of this issue - I'm not able to fix it. 
I can only do a blind guess and try Trial and error ...
The strangest thing is ... Everything is working in my test setup. 
I use a stock A1200 with no hardware extensions and clean (almost) OS.
So. 
I'm working on that, and trust me, it's it's terribly annoying. I'm Dreaming on that issue ... so it's serious.

gklka, 2023-03-01:
Don't take it too seriously :)

Is there a difference in mouSTer hardware? Or are all of the units behaving the same?

Maybe putting together a test configuration and a step-by-step instructions how to test it would help. It is clear that something is going off among us, and we are not testing the same thing, because it seems three different observations have been made:
- for you it works
- for some people it works on PAL but not on RTG
- for some neither PAL or RTG works

I'm glad to help you if I can.

midwan, 2023-03-01:
I agree that we need to be more specific with the test scenarios, in order to start making more sense of this and narrow things down.
In my case for example, all my systems have identical software installations: AmigaOS 3.2.1, P96 3.3.2, no hacks/patches on top if I can avoid it.

Some people mentioned it worked for them using other AmigaOS versions + FreeWheel + PAL.

Perhaps the OS version is related, as OS 3.2.x adds native support for the wheel and the FreeWheel tool is no longer required. If however the mouSTer only works with PAL and FreeWheel (and OS versions older than 3.2.x), that might explain some things.

@willyvmm 
Which OS are you testing with? I assume you're also running on PAL since you mentioned a standard A1200. Are you also using FreeWheel?

lipteklips2k15, 2023-03-01:
ok here are the systems I have tested on - A1200 with Vampire 1200, running Coffin OS R60, A1200 with TF1230 running Amiga OS 3.1, A500 with Firebird V4 running Coffin OS R61, A600 with PIStorm and Caffine OS and finally A1200 with PIStorm32 Lite and Caffine OS. The machines that have RTG have been tested using PAL and RTG screen modes.

midwan, 2023-03-01:
@lipteklips2k15 
CoffinOS and CaffeineOS are not really an OS however, it's just a packaged/bundled/patched together AmigaOS 3.9, which AFAIK can also be upgraded to 3.2.x. Best mention if you are using the 3.9 version or if you have done the upgrade to 3.2.x in this case.

You didn't mention if FreeWheel is being used?

Also, might be useful to mention the version of P96 used for the RTG modes. Not sure if it matters, but since this is still unknown it's best to gather all the details. :)

lipteklips2k15, 2023-03-01:
> @lipteklips2k15 CoffinOS and CaffeineOS are not really an OS however, it's just a packaged/bundled/patched together AmigaOS 3.9, which AFAIK can also be upgraded to 3.2.x. Best mention if you are [...]

Freewheel is installed on all systems and my actual Micromys adapter works fine on all setups including scroll wheel as for the OS installs of Caffine and Coffin they are as stock so thats 3.9 based (I wont pay for OS 3,2 for personal reasons) and the free version of P96.

gklka, 2023-03-01:
I just tested it with my Amiga 600, which has floppy only. Kickstart 2.05, booted from Workbench 3.1 Install floppy, then loaded the stable Micromys driver first, then FreeWheel 2.2.2 (from AmiNet) second from an another floppy.

I think it is also worth mentioning the programs you tried to scroll: I test with Workbench windows both in icon and list view, and ARTM which has a GadTools scroll widget.

And yeah, it did not work. :/

willyvmm, 2023-03-01:
Hi.
Ok.
Thanks for your support and willingness to help :)

I've missed something important or got fixed with some wrong assumption. 
So Let's take One small step at a time.

The first step is to confirm if there is no error in the USB descriptor parser engine.
Please install firmware from this link: https://we.tl/t-TiHytJm5kE
This firmware is based on the latest prerelease, and is preconfigured + has debug activated.
It should be fully functional, so there is nothing against using it :)

The procedure is simple:

1. Copy the file to the MOUSTER folder on a PenDrive.
2. Insert info the mouSTer
3. wait until it stops flashing rapidly, and start sending a morse R(remove) status
4. Remove the PendDrive
5. Connect your wheel mouse
6. Wait 1 second
7. Remove the Mouse
8. Insert PenDrive with an empty MOUSTER folder
9. Wait until it stops flashing rapidly, and start sending a morse R(remove) status
10. Remove the PenDrive, and send me the generated mouster.dbg file. Please attach a short info on what mouse it is.

In the 5th line, you should see **true** - that means that omuSTer has detected the wheel on your mouse.
Based on the debug data I will be able to partially emulate the mouse in my debug setup.

Looking forward to the results :)
Thanks.

willyvmm, 2023-03-01:
I have an Idea.

Could someone run Amiga Test Kit: https://github.com/keirf/Amiga-Stuff/releases
Then press F4, F1, F1, F1 (Controller ports and Change port 1 to Analog) 
And then use the wheel. This maybe sounds a bit stupid, but it's a very simple oscilloscope substitute.
You should be able to see something like this in this video: https://youtu.be/Xjt7pnYqhJA
The right dot is always ca in the same position, and the left one is changing position when using the wheel or press the middle button. Only one is visible on the screen at the same time.
The Bonus would be to take a video or few pictures (upload fx to https://wetransfer.com/)
The Supreme Bonus would be to do the same with Org. Micromys. (in this case a longer video would be appreciated)

Thanks for your support.

midwan, 2023-03-01:
@willyvmm 
Just did the test: https://youtu.be/7BhNGBFRIUk

- First tested the Micromys 5 adapter, with a PS2 mouse. The dot clearly moves when using the wheel or pressing it as a button.
- Then tested with all my USB mice (though the video only shows one, the behavior was the same with all). There is no response from the wheel moving, or being pressed at all.

midwan, 2023-03-02:
@willyvmm 
Attaching the debug output from fw `3.17.3475004` as well:
[MOUSTER.zip](https://github.com/willyvmm/mouSTer/files/10869181/MOUSTER.zip)

willyvmm, 2023-03-02:
Thanks. No error Here. 

I can state that all mouses has been recognized correctly. Its time to  get the next step.

gklka, 2023-03-02:
It seems, that M280 gives signals for scolling, but Performance MX is not:

M280: https://youtu.be/A-cva6UHLm0
Performance MX: https://youtu.be/ao96C5atA_M

willyvmm, 2023-03-02:
Thanks.
That's a huge help. Overall seems that I will write my own driver. The MM seems to be unnecessarily overcomplicated.
Additionally, the configuration process of the mouSTer firmware has to be simplified. 
Stay tuned.

JanNadziejowski, 2023-03-02:
Willy I checked this FW build 3.17.3475004. Unfortunately it does not work.

TomCat666, 2023-03-03:
I think your own driver might be the way to go. Also HID2AMI author wrote his own driver rather than use micromys since it was overcomplicated there too.

willyvmm, 2023-03-12:
This weekend I received a PiStorm32-Lite. It seems to be the most popular configuration.
And ... Everything is working. Workbench, Rtg. Classic, 3.2.1 ... I'm not able to recreate this issue.
It works every time.

gklka, 2023-03-12:
I just got my PiStorm32 Lite too, so I can test it soon in the next few days.

midwan, 2023-03-13:
I've also got a Pistorm32, so I did a test there today:
The scroll seems to work at first, but it's a bit unstable. Meaning, it scrolls in the window when you use the wheel, but then it also jumps/scrolls down if you move the mouse downwards, or click and drag the bottom right corner to resize a window, etc. Not always, but doing that a few times eventually triggers this.

It's always going down, never up.

TomCat666, 2023-03-13:
I have used it for around 2 weeks without any problems. Yesterday I upgraded to OS 3.2.2 and I also get this behaviour of mouse wheel events triggering just by moving the mouse and not touching the wheel. Doesn't happen all the time, but a lot of times, pretty much unusable.

willyvmm, 2023-03-15:
Thank YOU all for your support, patience and engagement. 

I have decided to drop support for the micromys protocol implementation and switch to a better suited digital protocol.

In short: **That was a mistake from my side. Simply lack of knowledge about Amiga hardware etc**.

From now the micromys support is considered **obsolete** and I will implement something new or proven. Likely something cocolino like.

I have created a new issue: #30 where the new protocol may be discussed. 
After succesful implementation the driver and the protocol will be published as an Open Source under non [VIRAL license](https://en.wikipedia.org/wiki/Criticism_of_copyleft)

closed 2023-03-15 by willyvmm

gklka, 2023-03-16:
Thank you for your efforts. I am a little bit late to the party, but I just wanted to add, that with PiStorm32 Lite and AmigaOS 3.2.2 I experience the same as @midwan and @TomCat666. The scroll kind of works but glitchy and scrolls down even when the wheel is not touched. I am waiting for the new driver.