willyvmm/mouSTer #29: Improved AUTO mode.

radius75, 2023-03-05:
Improved AUTO mode.
  Can you improve the automatic mode a bit? For example: When it detects a USB mouse, it will emulate one of the mouse modes selected in the INI (amiga, c1541, etc.)
  And connecting a USB gamepad to emulate one of the selected modes (gamepad, amigaCD32pad...)

  So that the user has the final influence on what he wants to run with such a configuration

willyvmm, 2023-03-05:
Hi @radius75 

Its already been partially implemented in the latest PreRelease [3.17.3475 - Pre-Release 3](https://github.com/willyvmm/mouSTer/releases/tag/3.17.3475)
This feature is not announced yet, because is not been fully tested. Feel free to test this feature, and send your suggestions.

In general, the config option: mode=[mouse|auto|gpad...] is going to be obsolete, and some config options will be rearranged. Also, some easy switching/config features will be added before the next stable release.

radius75, 2023-03-06:
Hello @willyvmm 

I suggest that the AUTO mode automatically selects the mouse that is indicated by the user in the [mouse]type=x option
And connecting a gamepad (arcdestick, joystick etc) to make a similar solution.
I use a bow tie once and a c64 gamepad once, each time I have to edit and load config.
This way, I would have one ini file for c64.
Automation in this direction is probably the most reasonable solution. A separate INI for each retrocomputer.
It will never be perfect.

willyvmm, 2023-03-06:
More-less that's my idea. 
I've just checked the status of this change and is not functional yet. Some changes are implemented in the 3475, but it's not enough.
So stay tuned.