willyvmm/mouSTer #30: New Wheel Protocol with Open Source Driver

willyvmm, 2023-03-15:
As the micomys protocol implementation is at least problematic (see here: #28 ) I have decided to drop the micromys protocol support and either implement one of existing or develop a new one.

willyvmm, 2023-03-17:
So.

Think twice.
Then.
Think twice.
And.
Code once.

The protocol is ready in my head. 

Stay tuned.

TomCat666, 2023-04-28:
For the new Tank Mouse a new open source driver was written for Mouse Wheel support that uses the cocolino protocol. It is working splendidly... maybe you could use that, no need to write a new driver then:

https://github.com/sq7bti/AmigaMouse/tree/tankmouse

TomCat666, 2023-11-06:
Anything new on this front ? :)

willyvmm, 2024-01-28:
The new protocol turned out to be more difficult than I thought. Mostly because of different limitations. 
Also, I want to push the Amiga hardware to its limits. Different hardware versions are limitations. 
I don't want different drivers for different hardware. 
But everything is on its way to a happy end.
I would say the most difficult part of the work seems to be done: [image]

willyvmm, 2024-02-07:
So. The driver is working. Still needs some work and polishing, but it is working :)

https://youtu.be/1TmlWyzdc14

Now, I can call myself CIA master ;)
I will write the whole story behind the driver soon.

willyvmm, 2024-02-18:
New firmware: https://github.com/willyvmm/mouSTer/releases/tag/3.20.4600
And Driver: https://github.com/willyvmm/Amiga-mouSTer-wheel-mouse-driver/releases/tag/0.8

So ... testing is needed.

gklka, 2024-02-20:
Hey, thanks for the great effort! My test:
- copied files to pendrive in MOUSTER folder
- plugged in, waited until .-. blink
- installed driver, rebooted
- plugged in mice:

Logitech Performance MX:
Wheel jumps up and down, but mostly down, basically unusable :( Even scrolls when I don't touch it.

Cheap Chinese USB/BT/2.4GHz mouse:
Wheel does not even move

Logitech M280:
Wheel does not even move

Cheap Chinese USB/2.4GHz vertical mouse:
Wheel does not even move

Logitech USB keyb+IR mouse:
Wheel does not even move

So it does not seem too good so far :(

willyvmm, 2024-02-27:
[mouSTer.fw.3.21.4660.zip](https://github.com/willyvmm/mouSTer/files/14425884/mouSTer.fw.3.21.4660.zip)
[mouSTer.driver_330.zip](https://github.com/willyvmm/mouSTer/files/14425920/mouSTer.driver_330.zip)

Completely new protocol. Slower but more reliable. The new driver, requires new firmware. Everything included. So far tested with my Stock A1200, and equipped with Pistorm32Lite with Pi3B(+). If tests are successful, I will copy everything to github.

gklka, 2024-02-29:
Tested all the mice again, and... outstanding success! All of them worked flawlessly. Congratulations, this implementation is great!

Thank you for all the hard work.

gklka, 2024-02-29:
Found a little glitch tho: keeping the mouse button pressed occasionally ends the event. For example when you drag a window around, in every 1-2 seconds it releases it.

lovrenco, 2024-03-07:
I confirm @gklka finding. When i drag dh0 icon on the desktop it changes place although button is not released

willyvmm, 2024-03-12:
[mouSTer.fw.3.21.4661.zip](https://github.com/willyvmm/mouSTer/files/14576636/mouSTer.fw.3.21.4661.zip)

Hi @lovrenco &  @gklka 
I have identified a bug in the firmware. Likely it's also responsible for the LMB issue. Could you please test the new 4661 firmware against the LMB issue?
As before, it's working on my hardware ... And I could not reproduce the reported issue.

gklka, 2024-03-15:
Works perfectly! The releasing mouse button bug is fixed, the scroll is still working like a charm. Thank you!

ales-zima, 2024-03-16:
Tried with A1200 and works with my wireless vertical mouse! Only issue I've found yet is DOpus4 - when scroll wheel is used, LMB stops working and starts again after few clicks. Anyway, thanks for great job!

jayminer81, 2024-09-12:
The link to the driver does not seem to be working any more :(

GT76a, 2024-10-01:
I have a problem with the adapter and wheel support. Tested with firmware [mouSTer.fw.3.21.4660.zip] and [mouSTer.fw.3.21.4661.zip]
 and driver [mouSTer.driver_330.zip]. I use a Logitech RX250 cable mouse. When I push the LMB the Mouster-Adapter stops/no heartbeat. The RMB and wheel is working until I press the LMB. Without the driver I the LMB is working fine, but no wheel support.

iamplouf, 2024-12-12:
Is there any description in wheel /4-5 prtitocol driver ?