willyvmm/mouSTer #31: Issue with cd32 joypad emulation in Flashback

moulder625, 2023-03-25:
Hi,

I've installed latest beta firmware and it has  solved ghost buttons press in most games for example Skeleton Crew is now playable. But in unnoficial CD32 Flashback port (with support for CD32 joypad) mouster is still pressing buttons by itself on splashscreens at the beginning with level, difficulty, language selection.

radius75, 2023-03-26:
To the best of my knowledge the original Flashback game is written for a Megadrive console (or other) gamepad. This game was not written for the amiga joyport standard.🤔

Retro-Nerd, 2023-03-26:
I've tested the Flashback WHDLoad with CD32 controls, latest WHDLoad slave v3.5. I used two different 8bitdo joypads. No issues with automatically pressed buttons. Mouster firmware is the latest beta.

moulder625, 2023-03-26:
There are issues in Unnoficial flashback port for cd 32 - http://unofficial-cd32-ports.blogspot.com/2018/07/006-flashback.html. Please check it out. I din't play whdload version.

Retro-Nerd, 2023-03-26:
I don't have a real CD32. But this version from Earok uses a special made WHDLoad CD32 version, with an older slave version of Psygore's patch (maybe 3.3). It's not an unofficial port of the game.  It's just the same the Amiga Disk version, patched for WHDLoad (with maybe a few tricks to make it work on a CD32 CD). 

I don't think there much to change here on the Mouster firmeware site.

willyvmm, 2023-11-06:
Not a mouSTer issue.

closed 2023-11-06 by willyvmm