willyvmm/mouSTer #32: Weird behavior after installing firmware pre-release 3.17.3475

WebFritzi, 2023-04-22:
Hi. Since my mouSTer (with cheap Logitech USB mouse) made some noise when plugged into a real C64, I decided to check out the latest pre-release 3.17.3475. So, I copied the ini-file to my harddrive, deleted it from the usb stick and copied the fw-file into the MOUSTER folder on my USB stick. Then plugged in the mouSTer into my C64, turned it on and then plugged in the USB drive. After that, I switched off the C64, deleted the fw-file from the USB drive and copied the ini-file back on there. Plugged it into the mouSTer with C64 on.

After this procedure, I supposed, my mouSTer would work just fine, but it did not. When I move the mouse, the mouSTer sends lots of signals to the C64, but not the correct ones. Lots of back-arrows appear (which it previously used to do only when I pressed mouse buttons). I tried to install the latest public firmware 3.16.2600, but that did not change anything. Now, my mouSTer does not work anymore. And, BTW, the noise is still there.

radius75, 2023-04-22:
For me, a wired mouse always introduced noise when moving. I use a wireless mouse, it causes minimal disturbance. They have to put on Volume Max on TV to hear it

radius75, 2023-04-22:
Inserting an empty pendrive and starting C64 does not save the INI file?

WebFritzi, 2023-04-22:
@radius75 Oh man!!! I did this and it worked! Thanks so much! But I wonder why. My old INI file worked before... Ok, and how do you configure wireless mice with the mouSTer? I just plugged in my wireless mouse and it didn't work.

radius75, 2023-04-22:
I tested many wired mice, each worked but had varying degrees of audible interference.

I have 3 Logitech wireless mice, they all work with Mouster.
You don't need to configure anything additionally. I connect the BT receiver to the Mouster and it works.

For example: Logitech M220, Logitech M235, the third is a very old Logitech mouse, no longer available for purchase ;)

It's probably a driver issue with your mouse.

My config for 1351 mouse, FW 3.17.3475 !!!

```
;
; 
; mouSTer - universal USB HID class device to db9 adapter. 
; Firmware revision: 3.17.3475

;
;                                          _/_/_/  _/_/_/_/_/                  
;    _/_/_/  _/_/      _/_/    _/    _/  _/            _/      _/_/    _/  _/_/
;   _/    _/    _/  _/    _/  _/    _/    _/_/        _/    _/_/_/_/  _/_/     
;  _/    _/    _/  _/    _/  _/    _/        _/      _/    _/        _/        
; _/    _/    _/    _/_/      _/_/_/  _/_/_/        _/      _/_/_/  _/         
;
;
; ###############################################################################
; #                                                                             #
; # Features:                                                                   #
; # Emulates Atari, Amiga, and Commodore 1351 mouse.                            #
; # Emulates standard Joystick, as well as Amiga CD32 Gamepad.
; # Can handle up to 16 buttons that can be translated to an action*            #
; # Can handle up to 6 axes, that can be translated to an action*               #
; #                                                                             #
; # More to come...                                                             #
; #         *an action - means any joystick direction or fire/mouse button      #
; #                                                                             #
; ###############################################################################
;
; ###############################################################################
; # Users Manual                                                                #
; ###############################################################################
;
; ###############################################################################
; # WARNING: WARRANTY MAY BE VOID IF USED WITH Atari 7800                       #
; ###############################################################################
; # Atari 7800 IS NOT SUPPORTED:                                                #
; # Please do not use the mouSTer with Atari 7800.                              #
; # It may be dangerous for the mouSTer due to weird A7800 construction.        #
; ###############################################################################
;
; ###############################################################################
; # Gamepad Buttons Mapping                                                     #
; ###############################################################################
; #  To make it easier, mouSTer is normalizing first 4 buttons of most gamepads #
; #  Default settings emulates only standard 1 button Joystick.                 #
; #              Button | SONY     | XBOX | Default Action                      #
; #        (4)        1 | SQUARE   | X    | Fire                                #
; #     (1)   (3)     2 | CROSS    | A    | Fire                                #
; #        (2)        3 | CIRCLE   | B    | AutoFire                            #
; #                   4 | TRIANGLE | Y    | AutoFire                            #
; ###############################################################################
;
; ###############################-Blink-Codes-###################################
;
; The mouSTer signals its state by using MORSE CODE, simply, blinking the LED.
; List of possible messages:
;
; Very fast blinking: 
; - Bootloader is Active and is waiting for the USB drive with new firmware.
; - If the USB drive is not detected within two seconds, 
[... 371 lines cut ...]
; [10|rb|fire] - axis is one direction button mapped to paddle A.
; [11|mb|fire] - axis is one direction button mapped to paddle B.

axis01=ns
axis02=ew
axis03=none
axis04=none
axis05=none
axis06=none

; validate=
; Validate if pressed direction button are giving valid joystick direction
; valid states are: n, ne, e, se, s, sw, w, nw or neutral
; invalid states are fx: ns, ew, new ,nesw etc.. When an invalid state is detected, the neutral state is applied
; buttons nor paddle never are validated.
; Default:
; validate=false
;
; Possible values:
; [00|false|off] - direction states are not validated.
; [01|true|on] - direction states are validated.

validate=false

```

radius75, 2023-04-22:
> EDIT: BTW, when I attach the (wired) mouse with the mouSTer to my Ultimate 64, there is only very little noise. Hardly hearable.

I think Ultimate64 has better filtering circuits for devices like modern mice.
In the days of C64, such filters were not provided.

WebFritzi, 2023-04-22:
Ok, thank you very much. I just ordered a Logitech M220 wireless mouse. Hope it works as it does for you.

closed 2023-04-22 by WebFritzi