willyvmm/mouSTer #36: Does not work well with "TheA500 Mini Mouse"

Mangerio, 2023-11-02:
TheA500 Mini Mouse works well with every PC.
Various Mouses work well with mouSTer.

TheA500 Mini Mouse does not work well with mouSTer.
mouSTer does not work well with TheA500 Mini Mouse.

The mouse pointer freezes often for a second.
As if the connection get's lost and must be reestablished
continously.

That's bad because TheA500 Mini Mouse is a good choice for Amigas!

willyvmm, 2023-11-06:
I have some thoughts on what it could be. As this is the ONLY mouse with that kind of issue, I have suspicions that it is not fully compatible with USB standards and is not able to handle high-rate pooling.
Unfortunately, I don't have "TheA500 Mini Mouse" to verify that, but I can build a test firmware to verify that.

Mangerio, 2023-11-06:
Thank you for your message!
It would be nice if there is a fix for it.
The A500 Mini Mouse is a nice small Tank Mouse.
Please let me know if you have a test firmware.
I will test it for you!

Mangerio, 2024-09-16:
The newest driver (3.20.4606 - Pre-Release 5) works better with the A500 Mini Mouse but the mouse pointer still freezes on a regular basis.

willyvmm, 2026-01-02:
Seems like this is the ONLY mouse that don't work correctly.

I have some ideas WHY - but it's only my assumptions. 
As for now, I can assume the problem is in the mouse itself, not in the mouSTer. 
Without the mouse and some extensive USB debugging, I'm not able to fix it, or rather create a workaround for this mouse.

I can't remember where, but as far as I remember, I read somewhere that this mouse has some other issues too.

Labryssa, 2026-09-22:
Oh no, this is ridiculous, I just got the mouSTer and went out of my way to buy this specific mouse 😭

I confirm that the mouse will stop working 50% of the time. It works, then it stops responding, you can do whatever you want the Amiga will not register your gesture. Then you jerk it and it works again.