willyvmm/mouSTer #37: Support for MSX / FM-Towns

ChrisTG742, 2024-01-24:
Are there any plans to support MSX mouse?
TIA!

willyvmm, 2024-01-28:
It is possible, but same as with SEGA different hardware is necessary or a kind of nasty adapter.

ChrisTG742, 2024-01-29:
Why that? AFAIK almost all legacy 9-pin controller ports operate with 5v TTL, either being high- or low-level. So the rest should be able to be done in software, no? Or am I missing something? Thanks!

willyvmm, 2024-01-29:
The problem is a different pinout. 
5V and GND are on different pins, and fixing these two pin locations in firmware is impossible.

ChrisTG742, 2024-01-31:
Ah, yes, of course. This could be solved by some jumpers or DIP-switches on the board maybe.

willyvmm, 2024-01-31:
Well ... In theory YES. But there are 2 pins to relocate to another location => total 4 pins. 
For Every pin, a 3-position jumper is needed.  Total: 12 positions. It's likely possible to optimize this to let's say 10 pins. 
10 pins take ca 30% PCB area of the mouSTer PCB.
So ... It's impossible and I will not consider this option in the future.

closed 2024-06-19 by willyvmm