willyvmm/mouSTer #38: Amiga middle and right button pull-up

gardners, 2024-01-28:
The Amiga right and middle buttons map to the POT lines on C64-compatible machines, including the MEGA65. On those machines, there is no pull-up line on the POT lines, which means that these buttons cannot be read.

A well-known work-around to this problem is to add a pull-up resistor internally in an Amiga mouse between 5V and the right mouse button line.

It would be great if the MouSTer had the option to emulate this for Amiga mouse mode, where it would drive the line high unless the mouse button is being pressed.

willyvmm, 2024-01-28:
add:

**revpotlines=true**

to your mouse config section.

Anyway, there is a new config system in my head. I will implement that when I release the new amiga protocol and driver.

gardners, 2024-01-29:
Thanks for the reply @willyvmm .  This option will invert the sense of the POT lines, right, rather than add the pull-up behaviour, until you do the new protocol etc?  That's fine. For the next MEGA65 core, we have put in logic that ignores the POT line if stuck low in Amiga mouse mode, so that revpotlines can be left false, and the mouse will still work with left-button only.  Let me know when you have the update, and I'll try it out.

willyvmm, 2024-01-30:
So ... lack of pull up/down resistor on the external port is just a design flaw. Quite annoying and kind of dangerous.
I can add a fix for the MEGA65 in the firmware (that is what exactly rovpotlines do, except without reversing the signal polarity).