willyvmm/mouSTer #44: Support for Atari MegaST/MegaSTE/TT ports

mikrosk, 2024-06-06:
I see that the official support is for Atari 260ST, 520ST, 1040ST(e) and Falcon. I have also noticed (with other USB adapters) that my MegaST/MegaSTE/TT keyboards indeed behave differently (buggy) when connected (one has to boot in such a bug state, then disconnect and reconnect them).

I didn't investigate further, is there something special about those keyboards?

willyvmm, 2024-06-19:
Hi @mikrosk  

Yes. It's kind of known issue. 
Due to the way how the keyboard is connected to the mainboard the 5V on the connector is dropping/unstable.
You may try to clean connectors using "contact spray" etc to improve mechanical contact.
In the very beginning of the mouSTer development i even build a special version dedicated to Mega series. It included an dedicated DC/DC converter to deliver stable 5V to USB.

mikrosk, 2024-06-19:
Bummer. :-/ I was hoping for some protocol change/different wiring issue but not this.

mikrosk, 2024-07-04:
Actually, there's more to it. I could reproduce the same problem with my Falcon, too. There it helped to press reset one more time and the adapter was good. I think this is related to the infamous reset problem on CT60, where it takes some time +5V to stabilise.

Now, why this doesn't help on MegaSTE/TT? Well... because those keyboards lack /RESET signal! So in whatever state they power on, they stay that way. So the only solution is to disconnect the keyboard, reconnect and *then* connect the adapter, as the +5V is then stable.

So two outcomes of this are:

- this is not an issue exclusively related to the MegaSTE/TT so perhaps you should include some kind of stabiliser (like e.g. this one for NetUSBee: https://www.atari-forum.com/viewtopic.php?p=436802#p436802)
- on MegaSTE/TT this has way more serious consequences as there is no way to reset IKBD except for disconnecting it

So maybe you can get the best out of it, make it bullet-proof with the stabiliser and get many MegaSTE/TT keyboards working at the same time?