willyvmm/mouSTer #48: 5-button joystick support for c64?

gurcei, 2025-08-27:
Hi there,
Was just wondering if it was possible to add support for this 5-button protocol for c64 joysticks?
- https://github.com/crystalct/5plusbuttonsJoystick

The technique [crystalct] used there was to do the following:
- button 1 = standard c64 fire button
- button 2 = POTX
- button 3 = POTY
- button 4 = up+down together
- button 5 = left+right together

I've tried your existing 3-button support for mouster, and that worked fine for me, when I configured things this way:
```
gbutton01=01  ;         normal c64 fire = my controller's X btn
gbutton02=02  ;         mouse right btn = my controller's A btn
gbutton03=03  ;           mouse mid btn = my controller's B btn
```

So maybe all the firmware would need is to add two new values for gbuttons, something like this maybe?
```
; [12|updown] - joystick up+down (aka 'SELECT' button in 5-button protocol)
; [13|leftright] - joystick left+right ((aka 'START' button in 5-button protocol)
gbutton13=12  ; my controller's START btn
gbutton14=13  ; my controller's BACK btn
```

Would that be feasible?

Regards,
Gurce

willyvmm, 2025-08-27:
To be honest, I have removed that feature a long time ago. I decided to Do Not support opposite directions at the same time :) 
So that should be quite easy to reimplement.
But the question is: Should I do that for a Single Person? Or are there more persons who would be interested?

gurcei, 2025-08-28:
How it has come about for me relates more to the MEGA65 presently.

Dan had a joystick tester program for the MEGA65 that is already aware of this 5-button protocol, so I was exploring configuring the mouster to suit it.

True, there are presently no c64 or mega65 games that support it, but having it available can at least open the possibility to develop multu-button games for our platform.

There can also be use cases for the MEGA65's arcade cores too, where we could make use of these extra two buttons for functionality like insert coin or start game.

So it would be nice to re-instate it, just to keep that option available to us developers.

I'm aware that there are potentially other multi-button protocols in the works (like protovision protopad).

While yes, sure, happy for support for such richer protocols in future, I reckon there's a quick win to be had with this simpler 5-button protocol too.

I'm iterating with one if the arcade core devs presently on another arcade core he's working on, and it could be a good chance for him to introduce these new buttons in that core.

gurcei, 2025-08-28:
Thank you, please keep me posted on where I can try the latest firmware that makes use of it, and I'll iterate with the arcade core dev and see if we can get it working.

gurcei, 2025-09-06:
Thank you, I'm giving it a try tonight.

One observation, I noticed that even in your latest "mouster.ini" file within the firmware zip package, it mentions the `revpotlines=` at two points in the file (line 218 and line 309). Was this intended for a reason, or just an accidental copy?

radius75, 2025-09-06:
One of them is for mouse. The other is for gamepad settings.

radius75, 2025-09-06:
Check the .msg text file for any errors while loading the .ini to the device. 
Or connect an empty flash drive and download the .ini from the device, to check if the configuration is correct.

gurcei, 2025-09-06:
Ah ok, got it now thanks, understood now about the two `revpotlines=`.

As for my probs with the non-working select/start, no probs, I figured out why it wasn't working.

It seems like in this newer firmware, the mapping of the gbuttons to my controller's buttons has changed slightly, and if I changed it to the following, it all now works :)

I.e. now I set gbutton09 = START and gbutton10 = SELECT.

```
gbutton01=01  ;         normal c64 fire = my controller's X btn
gbutton02=02  ;         mouse right btn = my controller's A btn
gbutton03=03  ;           mouse mid btn = my controller's B btn
gbutton04=*01 ; normal c64 fire (rapid) = my controller's Y btn
gbutton05=*02 ; mouse right btn (rapid) = my controller's LB btn
gbutton06=*03 ;   mouse mid btn (rapid) = my controller's RB btn
gbutton07=*02 ; mouse right btn (rapid) = my controller's LT btn
gbutton08=*03 ;   mouse mid btn (rapid) = my controller's RT btn
gbutton09=13  ; my controller's START btn
gbutton10=12  ; my controller's BACK btn
gbutton11=00  ;      (not used on my controller)
gbutton12=00  ;      (not used on my controller)
gbutton13=00  ;      (not used on my controller)
gbutton14=00  ;      (not used on my controller)
gbutton15=00  ; my controller's click of left-analog stick
gbutton16=00  ; my controller's click of right-analog stick
```

Great to see it working, I'll ping our arcade core dev, muse/shoestring and see if he'd like to try add support for this in his latest arcade cores and we'll see how it goes.

Thank you for the assist! :)

gurcei, 2025-09-16:
Ok, I've sent a PR to muse/shoestring to resolve this by making the jump button detection on potx/poty more lenient, so jump is working nicely once more with this newer mouster firmware and monsterland :-)

https://github.com/sho3string/SEGASYS1MEGA65_r3_r6/pull/2

With this tweak, the select and start button behaviour is working nicely now too, so thanks again  for the firmware update :-)

radius75, 2025-09-16:
This is a known issue; several people have reported it before. Some people also found the hex value <$10 unavailable when pressing the button. It all depends on the C64 you have. The only solution is to increase the tolerance range in the game/program code. Why not <$80? ;)

gurcei, 2025-09-24:
Just to note here, I've given the <$80 pressed and >=$80 released pathway a try and it has worked well in a tweaked Monsterland core. Muse is testing a few other multi-button joysticks he has, they've worked well also :)

gurcei, 2025-09-27:
Muse/shoestring has just updated his arcade cores for monsterland amd choplifter to now support 5-button protocol. He's working to do the same for choplifter next, so usage of the protocol is picking up momentum now, would be good to point the community to an official firmware release for their mousters that adds this support. Till then, I'll suggest them to use the interim firmware in this ticket.